autowarefoundation/AWSIM #370: can't start AWSIM Demo step by step according the AWSIM document

beforeface, 2024-12-05:
I successfully ran Autoware. But when I try to run ./AWSIM_v1.3.1/AWSIM.x86_64, I got the information as following:

Set current directory to /home/ggj
Found path: /home/ggj/AWSIM_v1.3.1/AWSIM.x86_64
no boot config - using default values

I did everything according the AWSIM document. How to solve it?

mackierx111, 2024-12-06:
@beforeface 
What errors occur when I activate AWSIM?
And what are your PC specs?

beforeface, 2024-12-08:
@mackierx111 
this  is the eror
[image]
This is my PC specs
[image]

taeseongkr, 2024-12-24:
@beforeface hello, did you solve that problem? I face the same issue

mackierx111, 2025-08-21:
Please try v2.0.0. If you encounter any problems, please create a new issue.
https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/

closed 2025-08-21 by mackierx111