autowarefoundation/AWSIM #381: Got a SIGSEGV while executing native code. This usually indicates a fatal error in the mono runtime or one of the native libraries  used by your application.

YAGbot, 2025-01-17:
$ nvidia-smi

+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.57.01              Driver Version: 565.57.01      CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3050 ...    On  |   00000000:01:00.0  On |                  N/A |
| N/A   39C    P0             12W /   60W |     920MiB /   4096MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1876      G   /usr/lib/xorg/Xorg                            402MiB |
|    0   N/A  N/A      2118      G   /usr/bin/gnome-shell                          118MiB |
|    0   N/A  N/A      3766      G   ...irefox/5437/usr/lib/firefox/firefox        317MiB |
|    0   N/A  N/A      5242      G   clash-verge                                     1MiB |
|    0   N/A  N/A      5366      G   ...gnu/webkit2gtk-4.0/WebKitWebProcess         18MiB |
+-----------------------------------------------------------------------------------------+

$ ./AWSIM.x86_64 
Set current directory to /home/aibot/autoware_simulation/awsim_ws/AWSIM_v1.3.1
Found path: /home/aibot/autoware_simulation/awsim_ws/AWSIM_v1.3.1/AWSIM.x86_64
Segmentation fault (core dumped) 

$ cd ~/.config/unity3d/TIERIV/AWSIM
$ gedit Player.log

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Caught fatal signal - signo:11 code:1 errno:0 addr:0xf0
Obtained 15 stack frames.
#0  0x007cb935a42520 in __sigaction
#1  0x007cb933011593 in (Unknown)
#2  0x007cb932f2a13d in (Unknown)
#3  0x007cb9345737d0 in __glx_Main
#4  0x007cb936966d4b in vk::AllocateBuffer(VkDevice_T*, VkPhysicalDeviceMemoryProperties const&, vk::MemoryAllocator&, VkBuffer_T*, unsigned int)
#5  0x007cb93699bad5 in vk::BufferManager::CreateBufferResource(unsigned long, unsigned int, unsigned int)
#6  0x007cb93699cc37 in vk::DataBuffer::CreateResource()
#7  0x007cb93699c91c in vk::DataBuffer::Initialize(void const*, vk::CommandBuffer*, vk::ScratchBuffer*, unsigned long)
#8  0x007cb93694b8db in GfxDeviceVK::InitializeBufferInternal(GfxBuffer*, void const*, GfxUpdateBufferFlags)
#9  0x007cb937439dc7 in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
#10 0x007cb93743accf in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
#11 0x007cb937431275 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
#12 0x007cb936d2149d in Thread::RunThreadWrapper(void*)
#13 0x007cb935a94ac3 in pthread_condattr_setpshared
#14 0x007cb935b26850 in __xmknodat

I did everything according the AWSIM document. How to solve it?

amkkashani, 2025-01-19:
Down grade your driver to 550 Nvidia, if problem persist converting the project to unity6 is a good idea for unity6 works well with awsim.

mackierx111, 2025-08-21:
Please try v2.0.0. If you encounter any problems, please create a new issue.
https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/