autowarefoundation/AWSIM #383: AWSIM v2.0.0 and new architecture

mackierx111, 2025-03-04:
Thank you for using AWSIM. Extensive refactoring and new architecture will be introduced to make AWSIM more scalable for development.

- I will create and publish an AWSIM design and coding convention that Guidelines to help developers better understand AWSIM.
- Refactor AWSIM as a whole to keep it consistent as a project; refine and abstract the functionality contained in AWSIM's main to make it more cross-cutting and usable. For example, some of the following will be implemented.
  - Support URP and HDRP in a single project, allowing users to choose whether rendering quality or computing performance is more important. Sample scenes of both are delivered. (In addition, unity after 6 [unified renderer](https://discussions.unity.com/t/unified-rendering/1519264) will be supported in the future.)
  - Allows entry points and main loops to be created for each scene in C# coding. Also avoids exploding references between Components in the UnityEditor's Inspector.
  - Remove `Awake()`, `Start()`, `Update()`, `FixedUpdate()`, etc. from each of AWSIM's MonoBehaviour components to allow explicit C# coding to control execution order without having to rely on "[Script Execution Order](https://docs.unity3d.com/Manual/class-MonoManager.html)."
  - Adopt a layered architecture in AWSIM and ｃlarify dependencies at the namespace level.
- Along with the refactoring, the entire document will be revamped.
- I will present best practices for forking or copying AWSIM to start a new project.

These tasks are done in a separate repository and then merged into the AWSIM main branch. Release date is still being worked out. Implementation is expected to take several months.

mackierx111, 2025-03-04:
Fewer commits may be added to the main branch until AWSIM v2.0.0.

sasa42, 2025-07-27:
@mackierx111 
The VehiclePhysicsPro VPP integration in awsim labs is an amazing helper to get better vehicle simulation. Would it be possible to add it in awsim too.

To suport of HDRP and URP is an great desiccision. We use booth for different use cases.

closed 2025-08-22 by mackierx111