autowarefoundation/AWSIM #401: Custom EgoVehicle

leonidaas, 2025-08-07:
I'm using **Version 2.0** of the project on **Ubuntu 22.04.**

After adding my custom **AccelVehicle** and following all the instructions as described in the documentation, the vehicle is not behaving as expected.

**Observed Issues**

    - The vehicle's speed is constantly fixed at 100 km/h, regardless of input.

    - Input controls are not working - even though I've configured everything the same way as with the Lexus vehicle.

    - Position tracking and sensor data appear to be working correctly.

**Additional Context**
If needed, I can provide my vehicle configuration files or share additional logs.

Please let me know if there’s anything I might have missed or if further debugging info would help.

leonidaas, 2025-08-07:
[ZOE_ACCEL_VEHICLE.zip](https://github.com/user-attachments/files/21652318/ZOE_ACCEL_VEHICLE.zip)

mackierx111, 2025-08-07:
[image]
Import failed. Please create a prefab that is not a variant.

leonidaas, 2025-08-07:
[ZOE_PREFAB.zip](https://github.com/user-attachments/files/21654262/ZOE_PREFAB.zip)

Maybe that works?

mackierx111, 2025-08-07:
Please include **all 3D models and materials referenced by the prefab**.
`ZOE_PREFAB.zip` only contains prefabs, so it is missing.

[image]

leonidaas, 2025-08-13:
I hope it works now.

mackierx111, 2025-08-21:
I'm sorry, but I couldn't download it. Could you please upload it again?

closed 2025-09-15 by leonidaas